BenMenking/routeros-api #3: PSR-2 compatibility

scottchiefbaker, 2015-08-31:
This library should be PSR-2 compatible for consistency. I see lots of coding sytles in here that should be standardized. Nothing major, just some code clean ups. PSR-2 requires PSR-1 compatibility, and PSR-1 says:

1. Class names MUST be declared in StudlyCaps.
2. Method names MUST be declared in camelCase.

This is only part of PSR-2 cleanup that will "break" things. How do we feel about this?

scottchiefbaker, 2015-08-31:
Here are the required changes as I see them:

1. Class name needs to change from routeros_api to RouterosAPI
2. Method is_iterable() needs to change to isIterable()
3. Method encode_length() needs to change to encodeLength()
4. Method parse_response() needs to change to parseResponse()
5. Method parse_response4smarty() needs to change to parseResponse4Smarty()
6. Method array_change_key_name() needs to change to arrayChangeKeyName()

Methods **not** requiring changes: `debug`, `connect`, `disconnect`, `read`, `write`, `comm`.

BenMenking, 2015-08-31:
Looks good.  I think some of the braces are not following PSR-2 as well as
method protection (public/private).

Ben Menking
*IT Security Consulting*
*ben@infotechsc.com <ben@infotechsc.com>*
Phone: 864.420.5603

scottchiefbaker, 2016-01-07:
This is done so I'm closing it.

closed 2016-01-07 by scottchiefbaker