BenMenking/routeros-api #6: SSL API with v6 RouterOS

scottchiefbaker, 2015-08-31:
This is an open issue to discuss doing API calls via SSL with newer versions of RouterOS. Is this something we can support?

BenMenking, 2015-08-31:
I think this is an absolute necessity

BenMenking, 2015-09-01:
I upgraded one of mine to v6.31 and will work on implementing api-ssl.  Pretty close already, but getting a weird SSL error.

scottchiefbaker, 2015-09-01:
I haven't even looked at the SSL API. What's involved in get PHP to talk to it?

BenMenking, 2015-09-01:
Should be as simple as changing the host in the fsockopen call to
"ssl://$hostname" as the fsockopen call will handle the SSL handshake and
negotiation.  Using a client certificate might require some additional work.

Ben Menking
*IT Security Consulting*
*ben@infotechsc.com <ben@infotechsc.com>*
Phone: 864.420.5603

scottchiefbaker, 2015-09-16:
Is there in any state I can test it?

BenMenking, 2015-09-21:
Sort-of.  I'm having problems getting it to work properly with my RouterOS.

Ben Menking
*IT Security Consulting*
*ben@infotechsc.com <ben@infotechsc.com>*
Phone: 864.420.5603

scottchiefbaker, 2016-01-07:
I'm just getting back to this, as we're getting really close to launching MikroTiks for our ISP customers. Where do we stand on API+SSL support?

shinradev, 2016-03-05:
I have created a pull request that would enable ssl functionality. Thanks.

lukemlsna, 2016-03-06:
@BenMenking Make sure you've specified a certificate under /IP/Services/ssl-api on your routeros device to properly enable encryption. If you are using a self-signed certificate, you _may_ need to instruct the client to trust that certificate. SSL connection worked for me in testing without having to import anything, but I used a publicly trusted cert. It also worked fine for me using a self-signed certificate on a debian cleint.

BenMenking, 2016-03-08:
@scottchiefbaker do these code changes help?

scottchiefbaker, 2016-03-08:
I haven't had a chance to test any of this. As long as we're making progress on it I'm happy.

exmac, 2016-03-10:
I was just test it, work Fine!

BenMenking, 2016-03-10:
Let's close this then.  If it becomes an issue, we can reopen.

closed 2016-03-10 by BenMenking