BenMenking/routeros-api #10: Call to undefined method RouterosAPI::parse_response()

src386, 2015-12-21:
Hi,
When I execute example1.php, I get results from the router so the connection is okay.
However, data is not parsed : 

    Call to undefined method RouterosAPI::parse_response()

Thanks.

exmac, 2015-12-21:
fixed on https://github.com/BenMenking/routeros-api/pull/11
or just use function: parseResponse

src386, 2015-12-21:
PHP Warning:  preg_match_all() expects parameter 2 to be string, array given in /home/xavier/routeros-api/routeros_api.class.php on line 171

(with ParseReponse).

exmac, 2015-12-21:
yes, looks like it was parsed by function read

exmac, 2015-12-21:
https://github.com/BenMenking/routeros-api/pull/12
check this

exmac, 2015-12-21:
in any way better use ->comm

src386, 2015-12-22:
I gave another try, it's working with parseReponse.
Thanks.

scottchiefbaker, 2016-01-07:
@src386 if this is working for you with the last commits please close this issue.

closed 2016-01-07 by src386