BenMenking/routeros-api #21: Is it possible to upgrade aunit via the API?

scottchiefbaker, 2016-09-08:
I'm attempting to upgrade a router via the API but I'm getting an error. `/system/package/update/check-for-updates` fetches information about the current and latest version. I can use that information to determine if there is an upgrade available. When I try and trigger an upgrade with `/system/package/update/upgrade` I get "no such command".

I can only assume one of two things:

1. I have the API call wrong
2. It's not possible to upgrade via the API

BenMenking, 2016-09-08:
I've always just FTP'd the upgrade files up to the mikrotik and rebooted.
It usually auto-finds the upgrade packages and installs.

Ben Menking
*IT Security Consulting*
*ben@infotechsc.com <ben@infotechsc.com>*
Phone: 864.420.5603

scottchiefbaker, 2016-09-08:
The auto upgrade works flawlessly via the CLI/SSH, but it doesn't appear to be available via the API. We've got about 800 of these deployed now, so an API method would be easiest.

BenMenking, 2016-09-08:
You may want to use /system/package/update/download

then follow that with a reboot command and then
/system/package/update/check-for-updates to verify it took.

Ben Menking
*IT Security Consulting*
*ben@infotechsc.com <ben@infotechsc.com>*
Phone: 864.420.5603

scottchiefbaker, 2016-09-08:
Ahhh good call. The command is different depending on version. 

    function needs_upgrade() {
        $cmd  = "/system/package/update/check-for-updates";
        $data = $this->api->comm($cmd);

        $cur_ver    = get_in($data,[1,'current-version']);
        $latest_ver = get_in($data,[1,'latest-version']);

        if ($cur_ver == $latest_ver) {
            return false;
        } else {
            return $data;
        }
    }

    function upgrade() {
        $needs_upgrade = $this->needs_upgrade();
        if ($needs_upgrade === false) {
            return false;
        }

        $cur_ver = floatval(get_in($needs_upgrade,[1,'current-version']));

        if ($cur_ver < 6.3) {
            $cmd = "/system/package/update/upgrade";
        } else {
            $cmd = "/system/package/update/install";
        }

        $data = $this->api->comm($cmd);

        return true;
    }

closed 2016-09-08 by scottchiefbaker