BenMenking/routeros-api #22: Cannot add new user for version 6.3++

adifaidz, 2016-11-21:
This code was working for v6.2, but for some reason, it didnt for 6.37. 
```
  $ARR=$this->API->createUserAndProfile(Array(
     "customer" => "admin",
     "name" => ("$username$postfix"),
     "password" => ("$username$postfix"),
     "numbers" => ("$username$postfix"),
     "profile" => $value,
  ));

  public function createUser(Array $data)
  {
    $this->comm("/tool/user-manager/user/add", $data);
  }

  public function createProfile(Array $data)
  {
    $this->comm("/tool/user-manager/user/create-and-activate-profile", $data);
  }

  public function createUserAndProfile(Array $data){
    $this->createUser(array_except($data, ['numbers', 'profile']));
    $this->createProfile(array_except($data, ['name', 'password']));
  }
```

This is the response message I got on 6.3+ was `unknown parameter`. Please help me in need of solution here 😢 🙏

Mikki-10, 2016-12-20:
Not sure what type of user that is you try to make?
But try:

$API->write('/user/add
							=name='.$username.'
							=password='.$password.'
							=group='.$group.'
							=address='.$address.'
							=disabled='.$disabled.'
							=comment='.$comment);

BenMenking, 2017-01-04:
Going to close this if it's not an issue....

closed 2018-03-01 by BenMenking