BenMenking/routeros-api #23: Lots of fwrite warnings in the latest version

scottchiefbaker, 2017-01-03:
I'm seeing a lot of this in my devel enviroment:

```
Warning: fwrite() expects parameter 1 to be resource, boolean given in /var/www/html/scott/rats/mikrotik-api/routeros_api.class.php on line 363

Warning: fwrite() expects parameter 1 to be resource, boolean given in /var/www/html/scott/rats/mikrotik-api/routeros_api.class.php on line 371
```

Not critical, but something that should be looked at.

BenMenking, 2017-01-03:
Looking at those lines, I'm suspecting that the socket is no longer valid (possibly false).  Ideally we would check for socket validity before the fwrite() and then handle any errors appropriately at that point.

Another possibility is that connect() was not called before write() is being called.  Regardless, write() should check for a valid socket and protest much if invalid.

scottchiefbaker, 2017-01-03:
I just realized that I was connecting to the wrong IP. In this case we were never able to connect at all... so we should add some tests that look for:

```
if (!$this->connected) {
    return false;
}
```

Before each call to read/write?

BenMenking, 2017-01-03:
Well, as I've been thinking about this, I'm wondering if we should also change connect() to only attempt once instead of multiple times.  Let the caller of connect() handle errors, which more than likely can't be fixed by multiple attempts.

Adding a socket validity check should be done.  Represent errors in the form of exceptions?

BenMenking, 2017-01-03:
Additionally, connect() should have returned false which the calling needs to trap and handle.  Maybe use an exception so it's pretty clear there was a fatal error.

scottchiefbaker, 2017-01-03:
I agree with everything you said. It's my fault for not handling `connect()` properly... but the library should also not toss a bunch of warnings if things go haywire.

BenMenking, 2017-01-03:
Fully agree.  I'll see what I can do here in the next hour or so.

scottchiefbaker, 2017-01-03:
For what it's worth... we've got over 1000 hEX's deployed now and we use this library to talk to all of them. It works great!

BenMenking, 2017-01-03:
Great to hear!  Maybe we can make some good incremental improvements during 2017 too :)

scottchiefbaker, 2017-01-03:
Now that we're relying on it more I'm sure I'll have some pull requests for you this year.

BenMenking, 2017-01-04:
I've made those changes in branch _better-error-handling_ so you can test them out.  Didn't go as far as adding in exceptions right now.

scottchiefbaker, 2017-01-04:
I don't think you pushed that branch. I'm not seeing that on my local repo, or on Github.

BenMenking, 2017-01-05:
It's there now.  Had to find my laptop :)

closed 2018-03-01 by BenMenking