BenMenking/routeros-api #24: how to print with where statement

irulvam, 2017-01-07:
i want to use this command by php
`ip hotspot active print detail  where user=username`

i try to use this code but i think  it wrong
`$ARRAY = $API->comm("/ip/hotspot/active/print/detail/where", array(
      "user" => "username",
   ));`

output message
`[message] => no such command prefix`

irulvam, 2017-01-08:
done. thanks
`$ARRAY = $API->comm("/ip/hotspot/active/print", array(
		"?user"=>"username",
   ));`

closed 2017-01-08 by irulvam

ve3, 2023-07-04:
If you would like to use command `where` with `or`, like this one in terminal.

```
/ip dhcp-server lease print where address=192.168.1.11 or address=192.168.1.12 or address=192.168.1.13
```

Here is sample code.

```php
$API->write('/ip/dhcp-server/lease/print', false);
$API->write('?address=192.168.1.11', false);
$API->write('?address=192.168.1.12', false);
$API->write('?#|', false);
$API->write('?address=192.168.1.13', false);
$API->write('?#|');
$result = $API->read();
// use `$result` variable.
```

See reference from manual [here](https://wiki.mikrotik.com/wiki/Manual:API#Query_word), and [here](https://wiki.mikrotik.com/wiki/Manual:API#Queries).