BenMenking/routeros-api #26: caps-man security password reset

dragos-grosu, 2017-02-10:
Trying to reset the password:

$API->write('/caps-man/security/set',false);
$API->write('=name="security-guest"=',false);
$API->write('=passphrase="g9rBxmkFvpPb"=');
$ARRAY = $API->read(false);
var_dump($ARRAY);
die();

I receive this response:
array(1) {
  [0]=>
  string(5) "!done"
}
But no password is reset.

In terminal, I run this command:
/caps-man security set security-guest passphrase=g9rBxmkFvpPb
and, as expected, it works. 

Not sure if it's a bug, or something I'm missing from the documentation.

BenMenking, 2017-02-14:
I tried to do the same thing with a ccr1019 but got the same results.  I don't know if this is bug in this API or Mikrotik's API.  It's not the first time a terminal command won't work using the MT API.   I suppose we'll leave it open for now.

BenMenking, 2018-03-01:
If this is still a problem, please reopen the issue.  Thanks!

closed 2018-03-01 by BenMenking