BenMenking/routeros-api #27: parseResponse removes wrong entries

sanglyb, 2017-09-27:
Hello, i've found a bug on class. The thing is - if there is a symbol = in response from api - parseResponse removes the line from the right after last symbol =.
Let me show you an exaple:
We want to get dns logs. The log itself looks like:
16:50:03 dns,packet id:f504 rd:1 tc:0 aa:0 qr:1 ra:1 QUERY 'no error' 
16:50:03 dns,packet question: freelancehunt.com:A:IN 
16:50:03 dns,packet answer: 
16:50:03 dns,packet <freelancehunt.com:A:119=104.25.147.98> 
16:50:03 dns,packet <freelancehunt.com:A:119=104.25.148.98> 
And after parseResponse [message] of the last 2 entries looks like:
<freelancehunt.com:A:119
<freelancehunt.com:A:119
Which is incorrect.

BenMenking, 2017-10-05:
Working on duplicating issue

BenMenking, 2019-07-08:
Can't duplicate

closed 2019-07-08 by BenMenking