BenMenking/routeros-api #33: parseResponse remove some of result

nozarc, 2018-07-14:
Hello, i have found trouble (or bug maybe) when parseResponse parsing result of my hotspot user profile on-login script
here the result before i parse it 
`=on-login=/system scheduler add name="remove-user-$user" interval="1d" on-even="/ip hotspot active remove [find user=$user];/ip hotspot user remove [find name=$user];/system scheduler remove [find name=remove-user-$user]"`
and after i parse it with parseResponse, parseResponse remove some of the result and it becomes like this
`[on-login] => /system scheduler add name`

EvilFreelancer, 2018-08-19:
@nozarc hello! You can try my edition of [ros api client](https://github.com/EvilFreelancer/routeros-api-php).

BenMenking, 2019-07-08:
Doesn't appear to be a bug?  Can't replicate.

closed 2019-07-08 by BenMenking