BenMenking/routeros-api #35: RouterOS API client on PHP7

EvilFreelancer, 2018-08-19:
Hello everybody! Do not think what is criticism, but this project has not been updated for many years and after a while it may stop working on a new versions of PHP.

Therefore, I created the same thing, like in this project but for PHP7 and in accordance with PSR standards. So probably [my project](https://github.com/EvilFreelancer/routeros-api-php) will be helpful for somebody.

PS. Sorry for this ADS and thanks to you guys for this elegant API client! Without your library, I would never understand how to work with the RouterOS API.

BenMenking, 2018-08-19:
What you should understand is that this library exists in this state for two reasons:

1) there are those who do not/can not use newer versions of PHP
2) there are already several other versions of this library for newer versions of PHP

Given that, you may want to get your project link on the Miktotik api Wiki page, which will have greater exposure than hijacking the github issue system.

closed 2018-08-19 by BenMenking

scottchiefbaker, 2018-08-20:
I use this project on PHP 7.1 without any issues.