BenMenking/routeros-api #37: How to add breaklines when Sent SMS API

muhammad-ihsan, 2018-12-10:
Good morning,
i have some problem adding breakline into my mikrotik API for sending a message. I already try to use `PHP_EOL`, `\n`, `\r\n`. There is no luck, the message only prints first line of the message.

Here is my code:
```
$mikrokit->comm("/tool/sms/send",array(
    "port"         => "usb1",
    "channel"      => "0",
    "phone-number" => $order->order_phone,
    "message"      => "[Rakki] Your order detail".PHP_EOL." Origin: ".$order->origin." ".PHP_EOL."Destination: ".$order->destination"
));
```

[image]

Chupaka, 2018-12-10:
Looks like the reason is here: https://github.com/BenMenking/routeros-api/blob/master/routeros_api.class.php#L371

**write** function splits each param by lines and sends them as separate commands :(

Chupaka, 2018-12-10:
Try to replace `$data = explode("\n", $command);` with `$data = [ $command ];` and recheck. If you're  not using `write` for sending multiple commands as one multiline string - it should work for you.

muhammad-ihsan, 2018-12-12:
@Chupaka  you are totally right. Thank you for the advice. I will recheck and try to fix this soon.

closed 2018-12-12 by muhammad-ihsan