BenMenking/routeros-api #39: openssl 1.1.0

eduardomazolini, 2019-03-06:
https://github.com/openssl/openssl/issues/8408
Change code for use ADH on new openssl.

$context = stream_context_create(array('ssl' => array('ciphers' => 'ADH:ALL:@SECLEVEL=0', 'verify_peer' => false, 'verify_peer_name' => false)));

BenMenking, 2019-07-08:
Since we want to maintain backwards compatibility (as much as possible), I don't think we want to make this change?  Anybody else care to comment on this?

scottchiefbaker, 2019-07-11:
I support moving forward and requiring new versions of SSL. Perhaps we can use a [check](https://stackoverflow.com/questions/9693614/how-to-check-if-installed-openssl-version-is-0-9-8k) to support the old version, but throw an `E_USER_DEPRECATED ` warning.