BenMenking/routeros-api #42: Problem with RouterOS version 6.45.1

michnzee, 2019-07-03:
Hello, after upgrade RouterOS 6.45.1 the PHP api class stop working because there is some changes in login procedure - via [https://wiki.mikrotik.com/wiki/Manual:API#Initial_login](url) - a lot of admin communities stop working their management scripts and it's reported in mikrotik official forum. It's possible to modify login procedure?
Log info: _system, error, critical: login failed for user myAPI from ipaddress via api._
Thank you

michnzee, 2019-07-03:
Aaahh... so, I upgrade latest version routeros-api class where is fix (api login above ros 6.43+) and there is all fixed. My shame :( Working good :)

ezechinwa, 2019-07-04:
> Aaahh... so, I upgrade latest version routeros-api class where is fix (api login above ros 6.43+) and there is all fixed. My shame :( Working good :)

Hi, Where can I find the router_api class for this new version

Chupaka, 2019-07-04:
> Hi, Where can I find the router_api class for this new version

It's right here, in this repo: https://github.com/BenMenking/routeros-api/blob/master/routeros_api.class.php

Chupaka, 2019-07-04:
> so, I upgrade latest version routeros-api class where is fix (api login above ros 6.43+) and there is all fixed
It was fixed a year ago, BTW :)
Please close the issue

closed 2019-07-04 by michnzee