BenMenking/routeros-api #43: New login method per 6.45.1

moudsen, 2019-07-08:
Remove all pre-6.45.1 code as the old login has been abandoned by Mikrotik.
https://wiki.mikrotik.com/index.php?title=Manual:API&oldid=33436#Initial_login

Find and replace the "function connect" to fix:

` public function connect($ip, $login, $password)
    {
        for ($ATTEMPT = 1; $ATTEMPT <= $this->attempts; $ATTEMPT++) {
            $this->connected = false;
            $PROTOCOL = ($this->ssl ? 'ssl://' : '' );
            $context = stream_context_create(array('ssl' => array('ciphers' => 'ADH:ALL', 'verify_peer' => false, 'verify_peer_name' => false)));
            $this->debug('Connection attempt #' . $ATTEMPT . ' to ' . $PROTOCOL . $ip . ':' . $this->port . '...');
            $this->socket = @stream_socket_client($PROTOCOL . $ip.':'. $this->port, $this->error_no, $this->error_str, $this->timeout, STREAM_CLIENT_CONNECT,
            if ($this->socket) {
                socket_set_timeout($this->socket, $this->timeout);
                $this->write('/login', false);
                $this->write('=name=' . $login, false);
                $this->write('=password=' . $password, true);
                $RESPONSE = $this->read(false);
                if (isset($RESPONSE[0]) && $RESPONSE[0] == '!done') {
                    $this->connected = true;
                    break;
                    }
                fclose($this->socket);
            }
            sleep($this->delay);
        }

        if ($this->connected) {
            $this->debug('Connected...');
        } else {
            $this->debug('Error...');
        }
        return $this->connected;
    }`

BenMenking, 2019-07-08:
We're going to need to maintain backward-compatibility for quite a while.  Maybe in 6 or 12 months this can be revisited?

closed 2019-07-08 by BenMenking

moudsen, 2019-07-08:
Well, your code no longer works at all on 6.45.1. Mikrotik has removed the challenge login method. I did check the current Git code and verified that this was the same code as I was using before.
I would agree to have a method in place to support both challenge and plain login methods, but as far as I can oversee the current code does not switch login modes if the challenge login fails ... Am I overseeing something?

moudsen, 2019-07-08:
I missed it ... (double checked and though I had the same code ... not!). Apologies for the fuzz!
Regards, Mark :-)

BenMenking, 2019-07-08:
Ok, so just for clarity, this code still works with 6.45.1?

moudsen, 2019-07-08:
Yep. Just replaced my home brew update with the one from the website and all is fine.

alvin-kiveu, 2022-08-17:
For 6.45.1 and above it does not work like for me I have tried in version 6.49.6 and it doesn't work please help.

moudsen, 2022-08-22:
Verified at my end: works like a charm on 6.49.6.
I have no opportunity nor any spare to test on 6.45.1. Strongly recommend to upgrade to 6.49.6 and test/try again or verify which routeros-api class version your are using? (I'm still on v1.6)

moudsen, 2022-08-22:
@BenMenking It's definitely way passed 6 to 12 months now ... LOL :-).

alvin-kiveu, 2022-08-22:
It is now okay