BenMenking/routeros-api #44: Access MK in Digital Ocean

dalcero, 2019-07-08:
Hello. As I access the Mikrotik with a Digital Ocean server, I have already released ports 8728 and 8729 and it still does not work .. It gives the error (Unable to establish socket session, Connection timed out)

BenMenking, 2019-07-09:
A couple things I would check:

Is api and/or api-ssl service enabled on mikrotik? 
Are there any mikrotik rules preventing connection to 8728 or 8729?
Are my digital Ocean access rules setup correctly?

tahsin352, 2019-07-14:
@dalcero have you setup iptables or firewall in your digital ocean server?? what do you mean by "already released ports 8728 and 8729 " ? 
"Connection timed out" means port is not enabled for access.
Is your mikrotik working? or it is a problem with API access to mikrotik?

rainiur, 2019-08-01:
api service is enabled and my server does not have a firewall. I am making a connection but unable to authenticate. I am able to login through the GUI with the username and password but unable to connect via the api. I am running 6.45.1. Any ideas?

BenMenking, 2019-08-01:
Can you provide more troubleshooting information?  What message does the script return?

wecc, 2019-11-02:
I'm not entirely sure but I think this issue might not be related to timeouts but instead caused by a breaking change in the login process (introduced in 6.45.1).

We also encountered issues being unable to login and after some troubleshooting, we discovered it was actually the login that failed and the multiple retries giving the impression of a timeout.

From the 6.45.1 release notes (https://mikrotik.com/download/changelogs):

> Important note!!! 
>Due to removal of compatibility with old version passwords in this version, downgrading to any version prior to v6.43 (v6.42.12 and older) will clear all user passwords and allow password-less authentication. Please secure your router after downgrading. 
Old API authentication method will also no longer work, see documentation for new login procedure: https://wiki.mikrotik.com/wiki/Manual:API#Initial_login 

The wiki link describes the changes in more detail.

I'd be happy to help to solve this issue, please let me know if I can provide any additional details.

pchf3788, 2019-11-25:
I hope the author updates API version, rookie learning!

bgf2, 2020-01-28:
yes that is because the API is updated in routeros but its not updated in this class yet.

Chupaka, 2020-01-30:
> but its not updated in this class yet.
WUT? It was updated more than a year ago!
https://github.com/BenMenking/routeros-api/pull/30

closed 2021-09-22 by BenMenking