BenMenking/routeros-api #52: Login failed using this routeros API.

debjit, 2020-11-08:
When trying to login using this script with example script I get an error that users cannot log in.

Router OS version:  6.47.6
user admin
all default value with admin/ password.

Here is what showing error PHP output:-
Connection attempt #1 to 192.168.0.101:8728... <<< [6] /login >>> [5/5] bytes read. >>> [5, 39]!done >>> [37/37] bytes read. >>> [37, 1]=ret=ee8d06ab50d811c5585b85ed6680f813 <<< [6] /login <<< [11] =name=admin <<< [44] =response=00e1c156cce9584df31d905a7b7e51c307 >>> [5/5] bytes read. >>> [5, 51]!trap >>> [42/42] bytes read. >>> [42, 8]=message=invalid user name or password (6) >>> [5/5] bytes read. >>> [5, 1]!done Connection attempt #2 to 192.168.0.101:8728... <<< [6] /login >>> [5/5] bytes read. >>> [5, 39]!done >>> [37/37] bytes read. >>> [37, 1]=ret=716166c3b79a21ff465d621341e027c0 <<< [6] /login <<< [11] =name=admin <<< [44] =response=00efb87bc3b99f46337208e5926621672a >>> [5/5] bytes read. >>> [5, 51]!trap >>> [42/42] bytes read. >>> [42, 8]=message=invalid user name or password (6) >>> [5/5] bytes read. >>> [5, 1]!done Connection attempt #3 to 192.168.0.101:8728... <<< [6] /login >>> [5/5] bytes read. >>> [5, 39]!done >>> [37/37] bytes read. >>> [37, 1]=ret=4921be73916280331b4baf6014287c43 <<< [6] /login <<< [11] =name=admin <<< [44] =response=00f23548e823be27d4f4d7b998b1c39741 >>> [5/5] bytes read. >>> [5, 51]!trap >>> [42/42] bytes read. >>> [42, 8]=message=invalid user name or password (6) >>> [5/5] bytes read. >>> [5, 1]!done Connection attempt #4 to 192.168.0.101:8728... <<< [6] /login >>> [5/5] bytes read. >>> [5, 39]!done >>> [37/37] bytes read. >>> [37, 1]=ret=b6e4ed085e839d0c0dd12587477cd15b <<< [6] /login <<< [11] =name=admin <<< [44] =response=0050672a24d04db9151164dbe797d46426 >>> [5/5] bytes read. >>> [5, 51]!trap >>> [42/42] bytes read. >>> [42, 8]=message=invalid user name or password (6) >>> [5/5] bytes read. >>> [5, 1]!done Connection attempt #5 to 192.168.0.101:8728... <<< [6] /login >>> [5/5] bytes read. >>> [5, 39]!done >>> [37/37] bytes read. >>> [37, 1]=ret=ddf457fb409973d1360ccab427c80991 <<< [6] /login <<< [11] =name=admin <<< [44] =response=003df34783a23f7dc61f3bfeb394f27064 >>> [5/5] bytes read. >>> [5, 51]!trap >>> [42/42] bytes read. >>> [42, 8]=message=invalid user name or password (6) >>> [5/5] bytes read. >>> [5, 1]!done Error... Disconnected...

error log from the router:
Topics: system
error
critical
message: login failure for user admin from 192.168.0.100 via API

debjit, 2020-11-08:
I really don't know what is happening but when I download the zip it gives me an old file. Sorry for this mishap.

closed 2020-11-08 by debjit

ve3, 2023-03-06:
Thank you.

I've installed via Composer but it give me old one. Download directly from **Code** > **Master branch** works now.

debjit, 2023-03-06:
Ok sure!