BenMenking/routeros-api #53: how to print torch data

fat-so, 2020-12-30:
i try to print torch data
i can successfully do this on mikrotik terminal
but i can't do this in api
here is my code 
$API->write('/tool/torch=interface=bridge1');
i'm not sure how to do this

Chupaka, 2021-01-04:
> $API->write('/tool/torch=interface=bridge1');

That command is completely invalid. It should be something like

```
$API->write('/tool/torch', false);
$API->write('=interface=bridge1');
```

Also, make sure you're reading a stream, not just waiting for the whole result, as this command returns the result infinitely (unless =duration=XXX is specified).

fat-so, 2021-01-08:
Thank you very much
and i want know how to print ip and macAddress on torch 
when i print the stream content , it just printed rx , tx , rx-packets , tx-packets and section number(the section number is useless , it's just like mikrotik make "section++")

fat-so, 2021-01-08:
> That command is completely invalid. It should be something like
>
> ```
> $API->write('/tool/torch', false);
> $API->write('=interface=bridge1');
> ```
>
> Also, make sure you're reading a stream, not just [...]
Thank you very much
and i want know how to print ip and macAddress on torch
when i print the stream content , it just printed rx , tx , rx-packets , tx-packets and section number(the section number is useless , it'

closed 2021-01-08 by fat-so

fat-so, 2021-01-08:
> That command is completely invalid. It should be something like
>
> ```
> $API->write('/tool/torch', false);
> $API->write('=interface=bridge1');
> ```
>
> Also, make sure you're reading a stream, not just [...]

is very helpful 
thx