BenMenking/routeros-api #58: torch not working

muhammadarsalansiddiqui, 2021-11-16:
this is my code and its not working.

        $API->write('/tool/torch',false);
        //$API->write("=interface=vlan3212");
        $API->write("=interface=".$rb_interface);
        $API->write("=src-address=".$rb_ip);
        $ARRAY= $API->read(false);
        print_r($ARRAY);

output:

Connection attempt #1 to 10.0.1.1:5891... <<< [6] /login <<< [13] =name=username_api<<< [26] =password=password_api >>> [5/5] bytes read. >>> [5, 1]!done Connected... <<< [11] /tool/torch <<< [19] =interface=vlan3212 <<< [25] =src-address=10.31.126.21 >>> [5/5] bytes read. >>> [5, 40]!trap >>> [31/31] bytes read. >>> [31, 8]=message=no such command prefix >>> [5/5] bytes read. >>> [5, 1]!done Array ( [0] => !trap [1] => =message=no such command prefix [2] => !done ) Disconnected... Disconnected...

Chupaka, 2021-11-16:
`$API->write("=interface=".$rb_interface);` should be `$API->write("=interface=".$rb_interface, false);`

muhammadarsalansiddiqui, 2021-11-16:
I also tries that, but it keeps the page loading.

```
        $API->write('/tool/torch');
        $API->write("=interface=NOC-40G",false);
        $ARRAY= $API->read(false);
        print_r($ARRAY);
```

Chupaka, 2021-11-16:
1) This should be
```
        $API->write('/tool/torch', false);
        $API->write("=interface=NOC-40G");
```
Are you sure you understand the second parameter?

2) Torch is constantly running command, it produces infinite output. If you need to stop at some point - add duration parameter, like
```
        $API->write('/tool/torch', false);
        $API->write("=interface=NOC-40G", false);
        $API->write("=duration=1s");
```

muhammadarsalansiddiqui, 2021-11-16:
i also tried to add duration parameter and interchanging false between these two statements

they return command does not exist

  1.  Torch is constantly running command, it produces infinite output. If you need to stop at some point - add duration parameter, like

Chupaka, 2021-11-16:
Please post exact code and the result.

muhammadarsalansiddiqui, 2021-11-17:
okay my problem is solved now, it was duration issue, 1s is not giving me results

muhammadarsalansiddiqui, 2021-11-17:
thank you for your support.

Chupaka, 2021-11-18:
Please close the issue

closed 2021-11-18 by muhammadarsalansiddiqui