BenMenking/routeros-api #63: Connecting to Mikrotik is  not work

alvin-kiveu, 2022-08-17:
I have tried to connect to the router using the API on Router OS v6.49.6 and it is not working is there an issue or this API  is outdated please help.

scottchiefbaker, 2022-08-17:
I've used this library to talk to v6.49+ and haven't had any issues. Can you post the code you're using to login?

alvin-kiveu, 2022-08-17:
Okay am using the code below to login

<?php

require('routeros_api.class.php');

$API = new RouterosAPI();

$API->debug = true;

if ($API->connect('192.168.88.1', 'admin', '12345678')) {

   $API->write('/interface/getall');

   $READ = $API->read(false);
   $ARRAY = $API->parseResponse($READ);

   print_r($ARRAY);

   $API->disconnect();
}

?>

I have connected to the  Mikrotik Router  and am using the same  as routeros_api.class.php in this repo but it is not working or there is a requirement I need to do.

wharjack, 2022-08-20:
Connection attempt #1 to 192.168.10.1:8728... Connection attempt #2 to 192.168.10.1:8728... Connection attempt #3 to 192.168.10.1:8728... Connection attempt #4 to 192.168.10.1:8728... Connection attempt #5 to 192.168.10.1:8728... Error... Disconnected...

Same

wharjack, 2022-08-20:
oh sorry I connected using the default config in winbox

BenMenking, 2022-08-22:
OK, considering this closed then.

closed 2022-08-22 by BenMenking