BenMenking/routeros-api #72: three bugs and their fixes in comm() function

volkirik, 2024-01-01:
first one:

`        $this->write($com, ($count===0));`

second (query commands need to be send with question mark)

```
                    case "?":
                        $el = "?$k=$v";
```

third (hang forever):

```
                $last = ($i === $count - 1);
                $this->write($el, $last);
				$i++;
```

Chupaka, 2024-01-02:
Any context here?..

What regards to

> case "?":
>                         $el = "?$k=$v";

It's `switch ($k[0])`, so your result will be something like `??key=val`

BenMenking, 2024-01-02:
@volkirik thank you for the report!  Can you provide any testing scenarios that can be run to duplicate the three issues?